astuff/pacmod_dbc #10: Flash File Version Number ID

asaxsma, 2019-04-09:
Requirement: Need way to identify code on PACMod component.
Message Name: FLASH_FILE_VERSION_XX
Message ID (Decimal): ???
Message ID (Hexadecimal): ???
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
~~Message Rate: 30 Hz~~
***Message Rate: 1 Hz***

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|MAJOR1|Unsigned|0|8|1|0|0|255||
|MINOR1|Unsigned|8|8|1|0|0|255||
|PATCH1|Unsigned|16|8|1|0|0|255||
|BRANCH2|Unsigned|24|8|1|0|0|255||
|CHANGE3|Unsigned|32|8|1|0|0|255||

1 Defined by Semantic Versioning 2.0.0.
2 Non-sequential number referencing a branch in version control.
3 Sequential number referencing a version on the branch.

asaxsma, 2019-05-16:
[Here](https://www.dropbox.com/s/qbze1veflcmz294/Firmware%20Identification.docx?dl=0) are the doc with our proposals.  Change tracking is on.  Feel free to elaborate, clarify, or add proposals.

asaxsma, 2019-05-21:
Guys, just noticed that nobody was assigned to this issue.  Hopefully you get the message this time.  Above is the proposal that we discussed.  Let's keep this discussion going so that we can nail this down.  Thanks!

JWhitleyAStuff, 2019-05-21:
I still think we should remove CHAR0...CHAR4 and replace it with a single field called something like "BUILD".

srustanAStuff, 2019-05-21:
> I still think we should remove CHAR0...CHAR4 and replace it with a single field called something like "BUILD".

Novel idea.

asaxsma, 2019-05-21:
Josh, you were favoring a mechanized way to pick out a pre-release number.  Are you still favoring that idea?

JWhitleyAStuff, 2019-05-21:
@asaxsma - The CAN data for those characters won't be human-readable anyway since they'll be ASCII-encoded We're just adding extra, unnecessary metadata by calling out each character individually. The PACMod driver can handle the decoding either way.

asaxsma, 2019-05-22:
We could use two bytes.  Every developer is assigned one of the numbers, and the other number is developer's choice.

JWhitleyAStuff, 2019-05-22:
@asaxsma - The RNG could be on each person's computer, no need to have any sort of network connection. The act of building would generate the random number and it would be built into the binary (and the filename). We can even do this in the CMake file: https://cmake.org/cmake/help/latest/command/string.html#generation

JWhitleyAStuff, 2019-05-22:
And if you want it to be as close to a true RNG as possible, we could use https://www.nist.gov/programs-projects/nist-randomness-beacon as a seed.

asaxsma, 2019-05-22:
The downside is for a 1 byte number, the likelihood of overlap is probably significant, even within the scope of a single MAJOR.MINOR.PATCH.  For a multiple byte number, it is probably insignificant, but it becomes somewhat unreadable.

JWhitleyAStuff, 2019-05-22:
I was talking about a 4-byte number.

asaxsma, 2019-05-22:
So here's an example of what that might look like:
1.1.5-0119197495
1.1.5-2422342601
1.1.5-0143941902
1.1.5-4120429306
1.1.5-0520388809

Hard to talk about.  Hard to distinguish one from another if numbers are close.  For example, (kind of a silly example, but makes the point I think)...
MikeL: "Hey Josh, what version of software did you leave on the machine last night".
JoshW: "Ah, oh yah, it was 1.1.5-01... something something.  Let me check."
MikeL: "No, that was from last week."
Josh: "No, its new, its just another -01 number.  Yes, it was 1.1.5-0119197495."
MikeL: "Say that again so that I can write it down."
JoshW: "0 - 1 - 1 - 9 - 1 - 9" take a breath "7 - 4 - 9 - 5".
MikeL: "Was that 191 or 919"?
JoshW: "Let me give it to you again.  0 - 1 - 1 - 9 - 1 - 9 - 7 - 4 - 9 - 5".
MikeL: "Got it.  I really think that was the number from last week.  Let me look it up."
JoshW:  Waits, tapping his fingers on his desk.
MikeL: "Okay, you are right.  Last week was an 011 number.  Thanks!"

JWhitleyAStuff, 2019-05-22:
And what you're talking about would be just as difficult:

JoshW: "Hey, Mike. I saw that 1.1.5-ML was left on the vehicle. Do you know what code was used to compile it?"
MikeL: *shrugs* "Its obviously mine but I made 30 builds yesterday. No idea."
JoshW: "So, is it safe to use?"
MikeL: "Better not. Just go back to the release."

I propose 2 letters (first/last) and 2 bytes of random numbers (0-65535):

1.1.5-JW27986
1.1.5-JW32847
1.1.5-JW02165

asaxsma, 2019-05-22:
How about this, since MikeL and JoshW are the only ones making code updates:
JoshW: "Hey Mike, gotta make some updates.  Guess we are on 1.1.6 now.  Should we just use "DEV" for suffixes again?"
MikeL: "Sounds good to me."
JoshW: releases 1.1.6-DEV00, spends the next couple days developing, and then releases 1.1.6-DEV01
MikeL: "Hey Mike, what number did you leave off with?"
JoshW: "01"
MikeL: Releases DEV02
A month goes by like that, producing:
1.1.6-DEV00
1.1.6-DEV01
1.1.6-DEV02
1.1.6-DEV03
1.1.6-DEV04

JWhitleyAStuff, 2019-05-22:
We agreed that this won't work. Too much coordination required between devs.

asaxsma, 2019-05-22:
Really, that was too much?

asaxsma, 2019-05-22:
I think Sam and I both shared that we've seen this work even at very large organizations.

asaxsma, 2019-06-07:
# Software Version Number
After much offline discussion, I believe we've arrived at this format:
|  For development: A-B  |  For release: C-D  |  Where:
A - Called "Software Version Number". Prior released Semantic Version 2.0 number.  Each of 3 digits is an integer [0-255].
B - Called "Software Build Number". 4 ASCII characters.  Each character is limited to [A-Z,0-9].  It is chosen randomly at build time.
C - Called "Software Version Number". New Semantic Version 2.0 number.   Each of 3 digits is an integer [0-255].
D - Called "Software Build Number". 4 ASCII characters "0000"

# Rational
1. We believe for any single Semantic version number, independent random number generators will not generate a duplicate number.  The chance of this is 1 in 1.6M (36^4).
2. We think this number is visually informative, clear, and easy to discuss.
3. We think representing the previous release with "A" is more informative because it provides information about the base release and it is cleaner because it does not attempt to predict the next release number, which could change.
4. In the VERY unlikely possibility that we make a release of an older release ("maintenance releases"), we think the Semantic version number is sufficient to trace this.

# Message Definition
Requirement: Need way to identify code on PACMod component.
Message Name: FLASH_FILE_VERSION_XX.  Where XX is 00, 01, 02, 03
Message ID (Decimal): 1032, 1033, 1034, 1035
Message ID (Hexadecimal): 0x408, 409, 40A, 40B
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 1 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|MAJOR<sup>1</sup>|Unsigned|7|8|1|0|0|255||
|MINOR<sup>1</sup>|Unsigned|15|8|1|0|0|255||
|PATCH<sup>1</sup>|Unsigned|23|8|1|0|0|255||
|BUILD0<sup>2</sup>|Unsigned|31|8|1|0|0|255|ASCII|
|BUILD1<sup>2</sup>|Unsigned|39|8|1|0|0|255|ASCII|
|BUILD2<sup>2</sup>|Unsigned|47|8|1|0|0|255|ASCII|
|BUILD3<sup>2</sup>|Unsigned|55|8|1|0|0|255|ASCII|

<sup>1</sup>MAJOR, MINOR, and PATCH shall be the software version number and shall be defined by Semantic Versioning 2.0.  For build numbers other than "0000", it shall be the most recent software release.
<sup>2</sup>BUILD0-BUILD3 shall represent the software build number and shall be constrained to characters A-Z and 0-9.

PR #28 (2019-06-25): Merge change 007
This closes #10, closes #11, and closes #21.
closed 2019-06-24 by asaxsma