astuff/pacmod_dbc #9: Component Message Watchdog Flag

asaxsma, 2019-04-09:
Requirement: Method for watchdog component to identify itself.
Message Name: COMPONENT_RPT_XX
Message ID (Decimal): 32
Message ID (Hexadecimal): 0x20
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|COMPONENT_TYPE1|Unsigned|0|4|1|0|0|15||
|COUNTER2|Unsigned|32|4|1|0|0|1||
|COMPLEMENT3|Unsigned|36|4|1|0|0|1||
|CONFIG_FAULT4|Unsigned|40|1|1|0|0|1||
|CAN_TIMEOUT_FAULT4|Unsigned|41|1|1|0|0|1||
|ESTOP5|Unsigned|42|1|1|0|0|1||
|ACCEL6|Unsigned|4|1|1|0|0|1||
|BRAKE6|Unsigned|5|1|1|0|0|1||
|CRUISE_CONTROL_BUTTONS6|Unsigned|6|1|1|0|0|1||
|DASH_CONTROLS_LEFT6|Unsigned|7|1|1|0|0|1||
|DASH_CONTROLS_RIGHT6|Unsigned|8|1|1|0|0|1||
|HAZARD_LIGHTS6|Unsigned|9|1|1|0|0|1||
|HEADLIGHT6|Unsigned|10|1|1|0|0|1||
|HORN6|Unsigned|11|1|1|0|0|1||
|MEDIA_CONTROLS6|Unsigned|12|1|1|0|0|1||
|PARKING_BRAKE6|Unsigned|13|1|1|0|0|1||
|SHIFT6|Unsigned|14|1|1|0|0|1||
|STEERING6|Unsigned|15|1|1|0|0|1||
|TURN6|Unsigned|16|1|1|0|0|1||
|WIPER6|Unsigned|17|1|1|0|0|1||
|***WATCHDOG6***|Unsigned|18|1|1|0|0|1||

1 0=PACMod, 1=PACMini, 2=PACMicro
2COUNTER shall have the value of 0 with the first message transmission.  It shall increase by 1 with each subsequent message transmission up to and including the value of 0xF.  The next message transmission shall be 0, and this pattern shall repeat.
3The COMPLEMENT shall be the complement of the COUNTER.  For example, if COUNTER is 0x1011, then the COMPLEMENT is 0x0100.
4 0=NO_FAULT, 1=FAULT
5 0=RELEASED, 1=PRESSED
6 0=ABSENT, 1=PRESENT

JWhitleyAStuff, 2019-04-09:
What's with the numbers after each signal name?

asaxsma, 2019-04-09:
They are footnotes.  Check out the template, its all explained there.

JWhitleyAStuff, 2019-04-09:
Ah.

asaxsma, 2019-04-09:
I'm not quite done with the template, but getting close I think.  I've set up Milestones, Labels, and a project also.  See what you think.  You can access the template as an administrator in the "Settings" tab, or as a user by just starting a new issue in the "Issues" tab.

commit 2a5bc6d1 (2019-05-21): Closes #9.  Closes #15.  Closes #19.
closed 2019-05-21 by driscoll85