astuff/pacmod_dbc #13: Publish DBC in HTML Format

asaxsma, 2019-04-24:
Automatically publish generated DBC in HTML format for readability and improved consistency.  This will replace the list of messages formally published in the PACMod 3.0 Startup Guide and more recently in the PACMod 3.0 General User Guide.

JWhitleyAStuff, 2019-04-24:
- [x] Update DBC converter tool to add to public vehicle_associations.xml instead of completely replacing it.
- [ ] Create command-line version of DBC converter tool to perform same functions as GUI version.
- [ ] Test auto-generation of general and vehicle-specific DBC HTML files.
- [ ] Add CI hooks to auto-generate and upload HTML documents to appropriate locations.

asaxsma, 2019-05-22:
We'll also need to update the generator at some point to include the new transmission type and frequency.  

Also, I'm wondering if we are specifying the frequency for periodic messages, do we need to specify the event for event based messages?

asaxsma, 2019-08-30:
Moved to BitBucket.

closed 2019-08-30 by asaxsma