astuff/pacmod_dbc #15: Message Transmission Rate

mlemm99, 2019-05-06:
## Transmission Rate Detail
Is there a place (dbc) or otherwise, we can document each messages' transmission rate?

asaxsma, 2019-05-06:
Guys, I was thinking that since the dbc is the reference, and I'm thinking there probably isn't a standard field for rate, maybe we could just put it in the message description field.  In the Vector db editor there is a place for a comment, and in the actual DBC it comes out as "CM_ BO_".

srustanAStuff, 2019-05-06:
Agreed.

JWhitleyAStuff, 2019-05-06:
@asaxsma / @srustanAStuff - there is a standard field for rate. It's a Message Attribute that you can add.

JWhitleyAStuff, 2019-05-06:
> @asaxsma / @srustanAStuff - there is a standard field for rate. It's a Message Attribute that you can add.

Sorry, it's not "standard." It's a user-defined attribute but it's available to add as a kind of "Default Attribute" in CanDBC++.

asaxsma, 2019-05-06:
Did a little research.  Here are some possible alternatives:
1.) Use message comments mentioned above.
2.) The dbc editor provides "TxMethod" and "Cycle Time" fields for a message.  I couldn't find any definitions for these.  TxMethod might relate to object oriented methods that can be set up in the Vector programming language CAPL, the Vector CANoe tool, or possibly other tools in the Vector tool chain.  Cycle time units are not defined, so I'm guessing they relate back to TxMethod.  There is some very limited information on this in Vector DBC++ help.
3.) We could define "user-defined attributes".  Apparently these can be whatever we want them to be, such as problematically accessed variables for tools we might want to create.  For example, I defined "Message_Period_ms" for a message object and assigned it to message "0" and gave it the value 1234.  Here's what it looks like in the DBC: BA_ "Message_Period_ms" BO_ 0 1234;.  There is some information on this in Vector DBC++ help.  Also, [here](https://assets.vector.com/cms/content/products/candb__/Docs/CANdb___Manual_EN.pdf) is a link.
4.) Vector seems to already have a set of predefined user-defined attributes (sounds like a oxymoron) for their tool chain.  Pertinent examples are: GenMsgSendType and GenSigSendType, GenMsgCycleTime and GenSigCycleTime.  There is some information on this in Vector DBC++ help.  Also, [here](https://github.com/stefanhoelzl/CANpy/blob/master/docs/DBC_Specification.md) a link.

JWhitleyAStuff, 2019-05-06:
@asaxsma - I'm pretty sure that GenMsgCycleTime is related to the TxMethod->Cycle Time field, but I'm not positive.

asaxsma, 2019-05-07:
I've got an email out to Vector to see if any of these options make sense or if there are other options.

asaxsma, 2019-05-16:
Team decided to use GenMsgCycleTime, GenMsgSendType, GenSigCycleTime, GenSigSendType.

commit 2a5bc6d1 (2019-05-21): Closes #9.  Closes #15.  Closes #19.
closed 2019-05-21 by driscoll85