astuff/pacmod_dbc #21: VIN, Firmware, Watchdog Report

mlemm99, 2019-05-22:
## VIN Report
Requirement: Add Transmission Rate info to DBC for VIN Report
Message Name: VIN Report
Message ID (Decimal): 1044
Message ID (Hexadecimal): 0x414
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 1 Hz (Proposed)

## Firmware Version Report
Requirement: Add Transmission Rate info to DBC for Firmware Version Report
Message Name: Firmware Version Report
Message ID (Decimal):
Message ID (Hexadecimal):
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 1 Hz (Proposed)

## Watchdog Report
Requirement: Add Transmission Rate and DBC info for Watchdog Report
Message Name: Watchdog Report
Message ID (Decimal): 1536 (Proposed)
Message ID (Hexadecimal): 0x600 (Proposed)
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 10 Hz (Proposed)

JWhitleyAStuff, 2019-05-23:
If we are going to make the Watchdog report a real part of the DBC, we should give it a "real" CAN ID, in accordance with https://github.com/astuff/pacmod_dbc/blob/v3/user_can_protocol.md. You OK with this, @mlemm99?

mlemm99, 2019-05-23:
Changing the CAN ID in firmware is easy. Updating the unknown number of Micros in customer hands will be harder.

We do have another change related to the notification command we could make in unison that would also require Micro updates. Perhaps we plan these changes together so only 1 Micro update needs to be made.

asaxsma, 2019-05-23:
What is a Watchdog Report?

asaxsma, 2019-06-11:
Merge Firmware version report request with related issue #10.  APPROVE.
Leave Vin report request.  APPROVE
Create new issue for the watchdog report.  Get message format defined.  Description about not for new development as a heartbeat message.

asaxsma, 2019-06-11:
Updated request:

VIN Report
Requirement: Add Transmission Rate info to DBC for VIN Report
Message Name: VIN Report
Message ID (Decimal): 1044
Message ID (Hexadecimal): 0x414
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 1 Hz (Proposed)

PR #28 (2019-06-25): Merge change 007
This closes #10, closes #11, and closes #21.
closed 2019-06-24 by asaxsma