astuff/pacmod_dbc #22: Define User Interaction

asaxsma, 2019-05-23:
Requirement: Don't understand what "USER_INTERACTION" means on most/all of the Aux messages.

JWhitleyAStuff, 2019-05-23:
`USER_INTERACTION` is any interaction on that system from the user's side. For steering, for example, this would mean that the user is currently moving the steering wheel. This applies whether the system is enabled, overridden, or disabled. It is useful for user software if they wish to have their own overrides (e.g. when `IGNORE_OVERRIDES = TRUE` on that system). It sometimes comes as feedback from the vehicle's CAN bus which we then forward on through PACMod.

JWhitleyAStuff, 2019-05-23:
Since we define `USER_PC` as the piece of hardware controlling the system through CAN, we may wish to call this `DRIVER_INTERACTION` instead.

asaxsma, 2019-05-24:
How's this for an update to the DBC?
Name: ~~USER_PC~~ DRIVER_INTERACTION
Values: 0=NO_INTERACTION, 1=INTERACTION
Description: DRIVER_INTERACTION shall have the value of 1 if the driver is moving, changing, or otherwise touching the operator control(s) that relates to this signal to an extent that is detectable.  Otherwise, the value shall be 0.

JWhitleyAStuff, 2019-05-28:
Do we use "DRIVER" in our startup guide? I think it's "OPERATOR," isn't it?

asaxsma, 2019-05-28:
Good catch.
Name: ~~USER_PC~~ OPERATOR_INTERACTION
Values: 0=NO_INTERACTION, 1=INTERACTION
Description: DRIVER_INTERACTION shall have the value of 1 if the driver is moving, changing, or otherwise touching the operator control(s) that relates to this signal to an extent that is detectable.  Otherwise, the value shall be 0.

JWhitleyAStuff, 2019-05-28:
I'm fine with the description too, provided we swap DRIVER for OPERATOR there.

asaxsma, 2019-05-28:
Name: USER_PC OPERATOR_INTERACTION
Values: 0=NO_INTERACTION, 1=INTERACTION
Description: OPERATOR_INTERACTION shall have the value of 1 if the driver is moving, changing, or otherwise touching the operator control(s) that relates to this signal to an extent that is detectable. Otherwise, the value shall be 0.

closed 2019-07-15 by mlemm99