astuff/pacmod_dbc #23: Component Report Update

mlemm99, 2019-05-25:
## PCB Voltage Rail Faults
We have hardware in place to monitor the voltage rails on PACMod and can easily add the functionality to PACMini in the future. These monitored voltages have not been used yet to detect board level or battery voltage level faults.

**Component Report ID**: 0x20 - 0x23
**Bit Position (proposed)**: 43
**Bit Length**: 1

|Byte\Bit|7|6|5|4|3|2|1|0|
|---|---|---|---|---|---|---|---|---|
|**0**|7|6|5|4|3|2|1|0|
|**1**|15|14|13|12|11|10|9|8|
|**2**|23|22|21|20|19|18|17|16|
|**3**|31|30|29|28|27|26|25|24|
|**4**|39|38|37|36|35|34|33|32|
|**5**|47|46|45|44|`43`|42|41|40|
|**6**|55|54|53|52|51|50|49|48|
|**7**|63|62|61|60|59|58|57|56|

JWhitleyAStuff, 2019-05-28:
@mlemm99 - Just to clarify, the request is just to add a single fault bit to the existing component reports, correct?

asaxsma, 2019-05-28:
I believe this is the request:

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|COMPONENT_TYPE 1|Unsigned|3|4|1|0|0|255||
|ACCEL 2|Unsigned|4|1|1|0|0|255||
|BRAKE 2|Unsigned|5|1|1|0|0|255||
|CRUISE_CONTROL_BUTTONS 2|Unsigned|6|1|1|0|0|255||
|DASH_CONTROLS_LEFT 2|Unsigned|7|1|1|0|0|255||
|DASH_CONTROLS_RIGHT 2|Unsigned|8|1|1|0|0|255||
|HAZARD_LIGHTS 2|Unsigned|9|1|1|0|0|255||
|HEADLIGHT 2|Unsigned|10|1|1|0|0|255||
|HORN 2|Unsigned|11|1|1|0|0|255||
|MEDIA_CONTROLS 2|Unsigned|12|1|1|0|0|255||
|PARKING_BRAKE 2|Unsigned|13|1|1|0|0|255||
|SHIFT 2|Unsigned|14|1|1|0|0|255||
|STEERING 2|Unsigned|15|1|1|0|0|255||
|TURN 2|Unsigned|16|1|1|0|0|255||
|WIPER 2|Unsigned|17|1|1|0|0|255||
|COUNTER a|Unsigned|35|4|1|0|0|15||
|COMPLEMENT b|Unsigned|39|4|1|0|0|15||
|CONFIG_FAULT 3|Unsigned|40|1|1|0|0|1||
|CAN_TIMEOUT_FAULT 3|Unsigned|41|1|1|0|0|1||
|ESTOP 4|Unsigned|42|1|1|0|0|1||
|**VOLTAGE_RAIL_FAULT 3**|**Unsigned**|**43**|**1**|**1**|**0**|**0**|**1**||

3 0=NO_FAULT, 1=FAULT

mlemm99, 2019-05-28:
yes, this looks correct

mlemm99, 2019-07-15:
Close #23

closed 2019-07-15 by mlemm99