astuff/pacmod_dbc #25: Command Limit Reports

cvigna9, 2019-06-06:
Command Limit Report Messages API request

Requirement: Adding message to display the maximum accepted accelerator command pacmod will execute when Command Limits are set. Also displays the users command.
Message Name: ACCEL_CMD_LIMIT_RPT
Message ID (Decimal): To be determined by committee. Currently 1344 
Message ID (Hexadecimal): To be determined by committee. Currently 0x540
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Startbit|Length|ValueType|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ACCEL_CMD_LIMIT|Unsigned|7|16|Motorola|0.001|0|0|1|none
|USER_ACCEL_CMD|Unsigned|23|16|Motorola|0.001|0|0|1|none

***BREAK

Requirement: Adding message to display the maximum accepted brake command pacmod will execute when Command Limits are set. Also displays the users command.
Message Name: BRAKE_CMD_LIMIT_RPT
Message ID (Decimal): To be determined by committee. Currently 1345
Message ID (Hexadecimal): To be determined by committee. Currently 0x541
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Startbit|Length|ValueType|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|BRAKE_CMD_LIMIT|Unsigned|7|16|Motorola|0.001|0|0|1|none
|USER_BRAKE_CMD|Unsigned|23|16|Motorola|0.001|0|0|1|none

****BREAK

Requirement: Adding message to display the maximum accepted steering commands for both position and rotation rate pacmod will execute when Command Limits are set. Also displays the users commands for position and rate.
Message Name: STEERING_CMD_LIMIT_RPT
Message ID (Decimal): To be determined by committee. Currently 1346
Message ID (Hexadecimal): To be determined by committee. Currently 0x542
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Startbit|Length|ValueType|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|MAX_POS_CMD_LIMIT|Signed|7|16|Motorola|0.001|0|-32.768|32.767|Rad
|USER_POS_CMD|Signed|23|16|Motorola|0.001|0|-32.768|32.767|Rad
|MAX_ROT_RATE_CMD_LIMIT|Unsigned|39|16|Motorola|0.001|0|0|32.768|Rad/s
|USER_MAX_ROT_RATE_CMD|Unsigned|55|16|Motorola|0.001|0|0|32.768|Rad/s

asaxsma, 2019-06-06:
@cvigna9 are we echoing the commands just to keep the command and its associated limit in synch with each other, or is there some other reason?

cvigna9, 2019-06-06:
No were echoing the original user command and limits that way it shows the user that they may be approaching the limit or if the command is above the limit it would show why whatever there commanding isnt being output.

asaxsma, 2019-06-07:
Maybe we could just use the command echo that is coming back in the system report and save the bits for something else?

JWhitleyAStuff, 2019-06-07:
I agree with @asaxsma - There is no real reason to duplicate these data in another message. I also highly recommend we only transmit these at 1Hz - it'll be completely static data - there is no reason to transmit them at a relatively high frequency.

JWhitleyAStuff, 2019-06-07:
I feel like maybe we should discuss having a request/response setup for messages like this, the VIN report, and the firmware version report. Thoughts from @asaxsma and @driscoll85?

asaxsma, 2019-06-07:
I spoke with Chris.  He's wanting the User PC to get 3 pieces of data: the unlimited command, the limited command, and the limits themselves.  The limits are dynamic based on various system variables.

cvigna9, 2019-06-07:
So i'll try to define it better. 

The command limits aren't static, they change based on variables such as time or speed. 

In the case of steering angle, steering max rotation rate, and accelerator commands the change according to vehicle speed and will always be dynamically changing. 

In the brakes case its based off of a timer that starts after the command goes above a threshold that allows the command to be incremented allowing safety driver more time to react in the case of inadvertent brake command being sent.   

I created these reports because if the command the user sends isnt above the limit then there is no way for the user to know what the limit at (x) speed is and how much "buffer" room they have before they reach the limit. 

I also thought it would be helpful to have them centralized for when developers are performing an analysis and to reduce the risk of data not lining up from the data changing between messages and the delay in one message to another. 

we can discuss more in the meeting.

asaxsma, 2019-06-20:
After some discussion, we believe there is 4 **distinct** pieces of information:
1. raw command: existing, useful for diagnosing communication w/ pacmod
2. limited command: new, useful for diagnosing communication w/ vehicle
3. limit: new, useful for control planning
4. reported command: existing, useful for diagnosing communication with vehicle

The proposed data link mapping (message / field):
1. raw command --> new limit report / user command
2. limited command --> existing system report / command
3. limit --> new limit report / command limit
4. reported command --> existing system report / output

Proposed change:
1. raw command --> **existing system report / command**
2. limited command --> **new limit report / user command**
3. limit --> new limit report / command limit
4. reported command --> existing system report / output

asaxsma, 2019-06-21:
Command Limit Report Messages API request

Requirement: Adding message to display the maximum accepted accelerator command pacmod will execute when Command Limits are set. Also displays the users command.
Message Name: ACCEL_CMD_LIMIT_RPT
Message ID (Decimal): To be determined by committee. Currently 513
Message ID (Hexadecimal): To be determined by committee. Currently 0x201
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Startbit|Length|ValueType|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|---|
|ACCEL_CMD_LIMIT|Unsigned|7|16|Motorola|0.001|0|0|1|none|
|**USER_ACCEL_LIMITED_CMD**|Unsigned|23|16|Motorola|0.001|0|0|1|none|

***BREAK

Requirement: Adding message to display the maximum accepted brake command pacmod will execute when Command Limits are set. Also displays the users command.
Message Name: BRAKE_CMD_LIMIT_RPT
Message ID (Decimal): To be determined by committee. Currently 517
Message ID (Hexadecimal): To be determined by committee. Currently 0x205
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Startbit|Length|ValueType|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|---|
|BRAKE_CMD_LIMIT|Unsigned|7|16|Motorola|0.001|0|0|1|none|
|**USER_BRAKE_LIMITED_CMD**|Unsigned|23|16|Motorola|0.001|0|0|1|none|

****Break

Requirement: Adding message to display the maximum accepted steering commands for both position and rotation rate pacmod will execute when Command Limits are set. Also displays the users commands for position and rate.
Message Name: STEERING_CMD_LIMIT_RPT
Message ID (Decimal): To be determined by committee. Currently 557
Message ID (Hexadecimal): To be determined by committee. Currently 0x22D
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Startbit|Length|ValueType|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|---|
|MAX_POS_CMD_LIMIT|Signed|7|16|Motorola|0.001|0|-32.768|32.767|Rad|
|**USER_POS_LIMITED_CMD**|Signed|23|16|Motorola|0.001|0|-32.768|32.767|Rad|
|MAX_ROT_RATE_CMD_LIMIT|Unsigned|39|16|Motorola|0.001|0|0|32.768|Rad/s|
|**USER_MAX_ROT_RATE_LIMITED_CMD**|Unsigned|55|16|Motorola|0.001|0|0|32.768|Rad/s|

asaxsma, 2019-06-25:
# Open Issues
I think we left some loose ends.  Please left me know if you agree to these:
1. I don't see ID's assigned.  I thought we decided to use the next values down in gaps between reports.  That would be accel: 0x201(513), brake: 0x205(517), and 0x22D (557).
2. I'm not sure why we agreed to e.g. "USER_ACCEL_LIMITED_CMD" when user "user" hasn't been used in association with a system command anywhere else.  What do you think about "ACCEL_CMD_AFTER_LIMIT"?  I think we said no to LIMITED_ACCEL_CMD.  I also considered ACCEL_CMD_LIMITED (too much like other field, ACCEL_CMD_LIMIT) and ACCEL_CMD_WITH_LIMIT (sound like 2 pieces of data, the cmd and the limit).
3. I was also thinking we were going to add descriptions.  How about these?
   1. ACCEL_CMD_LIMIT: "The ACCEL_CMD_LIMIT shall have the value of the limit imposed on the command associated with this system".
   2. ACCEL_CMD_AFTER_LIMIT: "The ACCEL_CMD_AFTER_LIMIT shall have the value of command associated with this system constrained by the ACCEL_CMD_LIMIT".
4. For the steering message, the names ought to reflect the associated command.  Agreed?
   1. POSITION_CMD_LIMIT
   2. POSITION_CMD_AFTER_LIMIT
   3. ROTATION_RATE_CMD_LIMIT
   4. ROTATION_RATE_CMD_AFTER_LIMIT

asaxsma, 2019-06-27:
Guys, still waiting on feedback from items 2-4 above.  Would prefer not to waste meeting time on this.

JWhitleyAStuff, 2019-06-27:
2. - I prefer "LIMITED_ACCEL_CMD" - not sure why we got rid of it. My second choice would be "ACCEL_CMD_LIMITED".
3. I'm OK with those descriptions.
4. Agreed.

driscoll85, 2019-06-27:
I approve, and agree with Josh on #2.

  1.

  2.  I'm OK with those descriptions.
  3.  Agreed.

asaxsma, 2019-06-27:
- Add comment on steering that the limit is absolute value.
- 2 above "LIMITED_ACCEL_CMD" with comment "The LIMITED_ACCEL_CMD shall be the value of the ACCEL_CMD of the associated system limited by the ACCEL_CMD_LIMIT".  
- 3 approved
- 4 agreed

closed 2019-08-30 by asaxsma