astuff/pacmod_dbc #27: Modify DLCs to Match Message Length

JWhitleyAStuff, 2019-06-11:
The majority of the messages in the PACMod DBC are only a few bytes but all messages are DLC = 8. This was for ease of programming but we are wasting an alarming amount of time on the bus for unused bytes. This issue is a request to reduce the DLC of each message to match its size.

asaxsma, 2019-06-12:
I'm wondering what this is going to break the interface for any of our customers?  First impression, customers aren't looking at DLC, except maybe to transfer data off of the CAN chip.  But, I wonder whether any of our customers' code has fixed this to 8 and will get a byte array either of the wrong size or with erroneous data.  Maybe for the same reason that we now have to make a code change.

JWhitleyAStuff, 2019-06-12:
I think the overwhelming majority of our customers are using either our ROS driver or something that uses `ros_canopen` or `kvaser_interface` for the raw data interpretation. If that isn't the case, we would find out really quickly. However, I really don't think we are affecting that many people by doing this - it'll be mostly the changes we have to make on our end which will still be minor.

closed 2019-08-30 by asaxsma