astuff/pacmod_dbc #30: Shift AUX Report

mlemm99, 2019-06-12:
## Overview
PACMini on Lexus uses the keypad to calibrate the Park position. When the Park position is recorded, the PACMini sets a flag for PACMicro so it knows to perform a triple beep to let the user know the Park position was recorded successfully.

This flag is undocumented and outside the future DLC range.

## Proposal
Move the bit inside the new DLC range and add it to the DBC

Message Name: SHIFT_AUX_RPT
Message ID (Decimal): 808
Message ID (Hexadecimal): 0x328
Frame Format: Standard
Data Length Code: 8 (future: 2)
TX Node: PACMini
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|WRITE_TO_CONFIG|Unsigned|12|1|1|0|0|1||

0=SILENT, 1=BEEP

asaxsma, 2019-06-13:
## Elaborate Request
Message Name: SHIFT_AUX_RPT
Message ID (Decimal): 808
Message ID (Hexadecimal): 0x328
Frame Format: Standard
Data Length Code: 8 (future: 2)
TX Node: PACMini
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|BETWEEN_GEARS|Unsigned|0|1|1|0|0|1||
|STAY_IN_NEUTRAL_MODE|Unsigned|1|1|1|0|0|1||
|BRAKE_INTERLOCK_ACTIVE|Unsigned|2|1|1|0|0|1||
|SPEED_INTERLOCK_ACTIVE|Unsigned|3|1|1|0|0|1||
|BETWEEN_GEARS_IS_VALID|Unsigned|8|1|1|0|0|1||
|STAY_IN_NEUTRAL_MODE_IS_VALID|Unsigned|9|1|1|0|0|1||
|BRAKE_INTERLOCK_ACTIVE_IS_VALID|Unsigned|10|1|1|0|0|1||
|SPEED_INTERLOCK_ACTIVE_IS_VALID|Unsigned|11|1|1|0|0|1||
|WRITE_TO_CONFIG<sup>1</sup>|Unsigned|12|1|1|0|0|1||

<sup>1</sup>0=SILENT, 1=BEEP

asaxsma, 2019-06-13:
I would propose the signal description:
<sup>a</sup>Notify the user that the shift system has permanently written data to its configuration file.

JWhitleyAStuff, 2019-07-11:
Closed by #42.

closed 2019-08-05 by asaxsma