astuff/pacmod_dbc #32: Add an extra Byte to send command to control XBR EBI-Mode, Priority and Control-Mode.

sanazfth, 2019-06-17:
Requirement: Add an extra Byte to send command to control XBR EBI-Mode, Priority and Control-Mode.
Message Name: External Brake Request Control Commands
Message ID (Decimal): 360
Message ID (Hexadecimal): 0x168
Frame Format: Standard
Data Length Code: 1
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|XBR_EBI_MODE|Unsigned|1|2|1|0|0|2||
|XBR_PRIORITY|Unsigned|3|2|1|0|0|3||
|XBR_CONTROL_MODE|Unsigned|5|2|1|0|0|2||
|NOT_DEFINED|Unsigned|7|2|1|0|3|3||

## Description

1
00=NO_ENDURANCE_BRAKE_INTEGRATION_ALLOWED
01=ONLY_ENDURANCE_BRAKES_ALLOWED
10=ENDURANCE_BRAKE_INTEGRATION_ALLOWED
11=NOT_AVAILABLE

2
00=HIGHEST_PRIORITY
01=HIGH_PRIORITY
10=MEDIUM_PRIORITY
11=LOW_PRIORITY

3
00=OVERRIDE_DISABLE
01=ACCELERATION_CONTROL_WITH_ADDITION_MODE
10=ACCELERATION_CONTROL_WITH_MAXIMUM_MODE
11-NOT_AVAILABLE

asaxsma, 2019-06-20:
There was some discussion that this deserves a new system.  Here is the consequent proposal.

Requirement: Add an extra Byte to send command to control XBR EBI-Mode, Priority and Control-Mode.

Message Name: BRAKE_DECCEL_CMD
Message ID (Decimal): 312
Message ID (Hexadecimal): 0x138
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLE|Unsigned|0|1|1|0|0|1||
|IGNORE_OVERRIDES|Unsigned|1|1|1|0|0|1||
|CLEAR_OVERRIDE|Unsigned|2|1|1|0|0|1||
|BRAKE_DECCEL_CMD|Unsigned|15|16|0.001|0|0|10|m/s^2|

Message Name: BRAKE_DECCEL_RPT
Message ID (Decimal): 568
Message ID (Hexadecimal): 0x238
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLED|Unsigned|0|1|1|0|0|1||
|OVERRIDE_ACTIVE|Unsigned|1|1|1|0|0|1||
|COMMAND_OUTPUT_FAULT|Unsigned|2|1|1|0|0|1||
|INPUT_OUTPUT_FAULT|Unsigned|3|1|1|0|0|1||
|OUTPUT_REPORTED_FAULT|Unsigned|4|1|1|0|0|1||
|PACMOD_FAULT|Unsigned|5|1|1|0|0|1||
|VEHICLE_FAULT|Unsigned|6|1|1|0|0|1||
|COMMAND_TIMEOUT|Unsigned|7|1|1|0|0|1||
|MANUAL_INPUT|Unsigned|15|16|0.001|0|0|10|m/s^2|
|COMMANDED_VALUE|Unsigned|31|16|0.001|0|0|10|m/s^2|
|OUTPUT_VALUE|Unsigned|47|16|0.001|0|0|10|m/s^2|

Message Name: **BRAKE_DECCEL_AUX_CMD**
Message ID (Decimal): **313**
Message ID (Hexadecimal): **0x139**
Frame Format: Standard
Data Length Code: **8**
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|XBR_EBI_MODE<sup>1</sup>|Unsigned|1|2|1|0|0|2||
|XBR_PRIORITY<sup>2</sup>|Unsigned|3|2|1|0|0|3||
|XBR_CONTROL_MODE<sup>3</sup>|Unsigned|5|2|1|0|0|2||
|NOT_DEFINED|Unsigned|7|2|1|0|3|3||
|**XBR_EBI_MODE_IS_COMMANDING<sup>4</sup>**|Unsigned|47|1|1|0|0|1||
|**XBR_PRIORITY_IS_COMMANDING<sup>4</sup>**|Unsigned|48|1|1|0|0|1||
|**XBR_CONTROL_MODE_IS_COMMANDING<sup>4</sup>**|Unsigned|49|1|1|0|0|1||
|**NOT_DEFINED_IS_COMMANDING<sup>4</sup>**|Unsigned|50|1|1|0|0|1||

<sup>1</sup> 00=NO_ENDURANCE_BRAKE_INTEGRATION_ALLOWED, 01=ONLY_ENDURANCE_BRAKES_ALLOWED, 10=ENDURANCE_BRAKE_INTEGRATION_ALLOWED, 11=NOT_AVAILABLE

<sup>2</sup> 00=HIGHEST_PRIORITY, 01=HIGH_PRIORITY, 10=MEDIUM_PRIORITY, 11=LOW_PRIORITY

<sup>3</sup> 00=OVERRIDE_DISABLE, 01=ACCELERATION_CONTROL_WITH_ADDITION_MODE, 10=ACCELERATION_CONTROL_WITH_MAXIMUM_MODE, 11-NOT_AVAILABLE

**<sup>4</sup> 0=IS_NOT_COMMANDING, 1=IS_COMMMANDING**

asaxsma, 2019-06-21:
Need further definition on requirements.

asaxsma, 2019-06-27:
Questions for Bendix:
1. What is the recommended priority?
2. Is it okay to expose the fact they they are using J1939 standard message (XBR) to command brake?

AS Question:
1. Should we allow access to all of these parameters?  Allow customer to break the vehicle.

Question to Customer:
1.  What are their control requirements?

asaxsma, 2019-06-28:
Pull out "not defined".
Remove is commanding.
Add vehicle feedback in aux message.
Make bits part of command message.

JWhitleyAStuff, 2019-06-28:
I approve with changes described by @asaxsma.

driscoll85, 2019-06-28:
I approve

asaxsma, 2019-07-22:
This issue continued on in #41.

closed 2019-08-30 by asaxsma