astuff/pacmod_dbc #34: Wheel Speed Report

mlemm99, 2019-06-20:
## Overview
Wheel Speed Report has 1 wheel set to unsigned and the other 3 wheels to signed

## Report
Message Name: WHEEL_SPEED_RPT
Message ID (Decimal): 1031
Message ID (Hexadecimal): 0x407
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

## Current
|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|WHEEL_SPD_FRONT_LEFT|Unsigned|7|16|0.01|0|-327.68|327.67|rad/s|
|WHEEL_SPD_FRONT_RIGHT|Signed|23|16|0.01|0|-327.68|327.67|rad/s|
|WHEEL_SPD_REAR_LEFT|Signed|39|16|0.01|0|-327.68|327.67|rad/s|
|WHEEL_SPD_REAR_RIGHT|Signed|55|16|0.01|0|-327.68|327.67|rad/s|

## Proposed
|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|WHEEL_SPD_FRONT_LEFT|Signed|7|16|0.01|0|-327.68|327.67|rad/s|
|WHEEL_SPD_FRONT_RIGHT|Signed|23|16|0.01|0|-327.68|327.67|rad/s|
|WHEEL_SPD_REAR_LEFT|Signed|39|16|0.01|0|-327.68|327.67|rad/s|
|WHEEL_SPD_REAR_RIGHT|Signed|55|16|0.01|0|-327.68|327.67|rad/s|

JWhitleyAStuff, 2019-06-21:
Wow. I can't believe we missed this until now.

asaxsma, 2019-06-21:
I'm guessing its supposed to be signed.  Is that correct?

JWhitleyAStuff, 2019-06-21:
It is, yes.

JWhitleyAStuff, 2019-07-11:
Closed by #42.

closed 2019-07-11 by JWhitleyAStuff