astuff/pacmod_dbc #35: Wiper Command

mlemm99, 2019-06-21:
## Overview
Add 1 more intermittent level to support more platforms.
Add sprayer control

## Wiper Command
Message Name: WIPER_CMD
Message ID (Decimal): 308
Message ID (Hexadecimal): 0x134
Frame Format: Standard
Data Length Code: 8 (Future: 3)
TX Node: PACMOD
Message Rate: 30 Hz

## Current
|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLE|Unsigned|0|1|1|0|0|1||
|IGNORE_OVERRIDES|Unsigned|1|1|1|0|0|1||
|CLEAR_OVERRIDE|Unsigned|2|1|1|0|0|1||
|WIPER_CMD  **1**|Unsigned|15|8|1|0|0|1||

**1**
7=High, 6=Low, 5=Intermittent 5, 4=Intermittent 4, 3=Intermittent 3, 2=Intermittent 2, 1=Intermittent 1, 0=Wipers Off

## Proposed
|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLE|Unsigned|0|1|1|0|0|1||
|IGNORE_OVERRIDES|Unsigned|1|1|1|0|0|1||
|CLEAR_OVERRIDE|Unsigned|2|1|1|0|0|1||
|WIPER_CMD  **1**|Unsigned|15|8|1|0|0|1||
|SPRAYER_CMD  **2**|Unsigned|16|1|1|0|0|1||

**1**
8=High, 7=Low, 6=Intermittent 6, 5=Intermittent 5, 4=Intermittent 4, 3=Intermittent 3, 2=Intermittent 2, 1=Intermittent 1, 0=Wipers Off

**2**
0 = NOT SPRAYING, 1 = SPRAYING

JWhitleyAStuff, 2019-06-21:
So, this is purely because we have a "defined architecture" for system types (for example, the WIPER_CMD is an "IntSystem," I would prefer to have the SPRAYER_CMD be another system (BoolSystem). It avoids coding a special-case for the Wiper system in the firmware and the driver. Currently STEER_CMD is the only special-case system because it needs both the position and rotation rate in order to work properly. Do you have any problems with this?

JWhitleyAStuff, 2019-06-24:
Closing in favor of #37.

mlemm99, 2019-06-24:
add intermittant level

asaxsma, 2019-06-27:
255 - HIGH, 254 - MEDIUM, 253 - LOW, 11 - INTERMITTENT10 (fastest) through INTERMITTENT1 (slowest), 0 - OFF

JWhitleyAStuff, 2019-07-11:
Closed by #42.

closed 2019-08-05 by asaxsma