astuff/pacmod_dbc #38: Component Report - Add Spray System Flag

mlemm99, 2019-06-24:
## Overview
Requirement: Add Spray System flag to component report systems.
Message Name: COMPONENT_RPT_00-03
Message ID (Decimal): 32 - 35
Message ID (Hexadecimal): 0x20 - 0x23
Frame Format: Standard
Data Length Code: 8 (Future: 6)
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|COMPONENT_TYPE 1|Unsigned|3|4|1|0|0|15||
|ACCEL 2|Unsigned|4|1|1|0|0|1||
|BRAKE 2|Unsigned|5|1|1|0|0|1||
|CRUISE_CONTROL_BUTTONS 2|Unsigned|6|1|1|0|0|1||
|DASH_CONTROLS_LEFT 2|Unsigned|7|1|1|0|0|1||
|DASH_CONTROLS_RIGHT 2|Unsigned|8|1|1|0|0|1||
|HAZARD_LIGHTS 2|Unsigned|9|1|1|0|0|1||
|HEADLIGHT 2|Unsigned|10|1|1|0|0|1||
|HORN 2|Unsigned|11|1|1|0|0|1||
|MEDIA_CONTROLS 2|Unsigned|12|1|1|0|0|1||
|PARKING_BRAKE 2|Unsigned|13|1|1|0|0|1||
|SHIFT 2|Unsigned|14|1|1|0|0|1||
|STEERING 2|Unsigned|15|1|1|0|0|1||
|TURN 2|Unsigned|16|1|1|0|0|1||
|WIPER 2|Unsigned|17|1|1|0|0|1||
|WATCHDOG 2|Unsigned|18|1|1|0|0|1||
|SPRAY 2|Unsigned|?|1|1|0|0|1||
|COUNTER a|Unsigned|35|4|1|0|0|15||
|COMPLEMENT b|Unsigned|39|4|1|0|0|15||
|CONFIG_FAULT 3|Unsigned|40|1|1|0|0|1||
|CAN_TIMEOUT_FAULT 3|Unsigned|41|1|1|0|0|1||
|ESTOP 4|Unsigned|42|1|1|0|0|1||

1
0=PACMod, 1=PACMini, 2=PACMicro

2
0=ABSENT, 1=PRESENT

3
0=NO_FAULT, 1=FAULT

4
0=RELEASED, 1=PRESSED

a
Counter shall have the value of 0 with the first message transmission. It shall increase by 1 with each subsequent message transmission up to and including the value of 0xF. The next message transmission shall be 0, and this pattern shall repeat.

b
The COMPLEMENT shall be the complement of the COUNTER. For example, if COUNTER is 0x1011, then the COMPLEMENT is 0x0100.

asaxsma, 2019-06-27:
Add a bit.  ESTOP_FAULT.
Sprayer bit okay as defined.

JWhitleyAStuff, 2019-07-11:
Closed by #42.

closed 2019-08-05 by asaxsma