astuff/pacmod_dbc #39: Spray System Report

mlemm99, 2019-06-24:
## Template
Requirement: Add message report for Spray System
Message Name: SPRAYER_RPT
Message ID (Decimal):
Message ID (Hexadecimal):
Frame Format: Standard
Data Length Code: 8 (Future: 4)
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLED|Unsigned|0|1|1|0|0|1||
|OVERRIDE_ACTIVE|Unsigned|1|1|1|0|0|1||
|COMMAND_OUTPUT_FAULT|Unsigned|2|1|1|0|0|1||
|INPUT_OUTPUT_FAULT|Unsigned|3|1|1|0|0|1||
|OUTPUT_REPORTED_FAULT|Unsigned|4|1|1|0|0|1||
|PACMOD_FAULT|Unsigned|5|1|1|0|0|1||
|VEHICLE_FAULT|Unsigned|6|1|1|0|0|1||
|COMMAND_TIMEOUT 1|Unsigned|7|1|1|0|0|1||
|MANUAL_INPUT|Unsigned|8|8|1|0|0|1||
|COMMANDED_VALUE|Unsigned|16|8|1|0|0|1||
|OUTPUT_VALUE|Unsigned|24|8|1|0|0|1||

asaxsma, 2019-06-24:
Maybe next time around combine all this into a single issue.

JWhitleyAStuff, 2019-07-11:
Closed by #42.

closed 2019-08-05 by asaxsma