astuff/pacmod_dbc #41: Add an extra Byte to send command to control XBR EBI-Mode, Priority and Control-Mode.

asaxsma, 2019-06-28:
Requirement: Add an extra Byte to send command to control XBR EBI-Mode, Priority and Control-Mode.
Message Name: External Brake Request Control Commands
Message ID (Decimal): 360
Message ID (Hexadecimal): 0x168
Frame Format: Standard
Data Length Code: 1
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|XBR_EBI_MODE|Unsigned|1|2|1|0|0|2||
|XBR_PRIORITY|Unsigned|3|2|1|0|0|3||
|XBR_CONTROL_MODE|Unsigned|5|2|1|0|0|2||
|NOT_DEFINED|Unsigned|7|2|1|0|3|3||

## Description

1
00=NO_ENDURANCE_BRAKE_INTEGRATION_ALLOWED
01=ONLY_ENDURANCE_BRAKES_ALLOWED
10=ENDURANCE_BRAKE_INTEGRATION_ALLOWED
11=NOT_AVAILABLE

2
00=HIGHEST_PRIORITY
01=HIGH_PRIORITY
10=MEDIUM_PRIORITY
11=LOW_PRIORITY

3
00=OVERRIDE_DISABLE
01=ACCELERATION_CONTROL_WITH_ADDITION_MODE
10=ACCELERATION_CONTROL_WITH_MAXIMUM_MODE
11-NOT_AVAILABLE

asaxsma, 2019-06-28:
There was some discussion that this deserves a new system.  Here is the consequent proposal.

Requirement: Add an extra Byte to send command to control XBR EBI-Mode, Priority and Control-Mode.

Message Name: BRAKE_DECCEL_CMD
Message ID (Decimal): 312
Message ID (Hexadecimal): 0x138
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLE|Unsigned|0|1|1|0|0|1||
|IGNORE_OVERRIDES|Unsigned|1|1|1|0|0|1||
|CLEAR_OVERRIDE|Unsigned|2|1|1|0|0|1||
|BRAKE_DECCEL_CMD|Unsigned|15|16|0.001|0|0|10|m/s^2|

Message Name: BRAKE_DECCEL_RPT
Message ID (Decimal): 568
Message ID (Hexadecimal): 0x238
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLED|Unsigned|0|1|1|0|0|1||
|OVERRIDE_ACTIVE|Unsigned|1|1|1|0|0|1||
|COMMAND_OUTPUT_FAULT|Unsigned|2|1|1|0|0|1||
|INPUT_OUTPUT_FAULT|Unsigned|3|1|1|0|0|1||
|OUTPUT_REPORTED_FAULT|Unsigned|4|1|1|0|0|1||
|PACMOD_FAULT|Unsigned|5|1|1|0|0|1||
|VEHICLE_FAULT|Unsigned|6|1|1|0|0|1||
|COMMAND_TIMEOUT|Unsigned|7|1|1|0|0|1||
|MANUAL_INPUT|Unsigned|15|16|0.001|0|0|10|m/s^2|
|COMMANDED_VALUE|Unsigned|31|16|0.001|0|0|10|m/s^2|
|OUTPUT_VALUE|Unsigned|47|16|0.001|0|0|10|m/s^2|

Message Name: **BRAKE_DECCEL_AUX_CMD**
Message ID (Decimal): **313**
Message ID (Hexadecimal): **0x139**
Frame Format: Standard
Data Length Code: **8**
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|XBR_EBI_MODE<sup>1</sup>|Unsigned|1|2|1|0|0|2||
|XBR_PRIORITY<sup>2</sup>|Unsigned|3|2|1|0|0|3||
|XBR_CONTROL_MODE<sup>3</sup>|Unsigned|5|2|1|0|0|2||
|NOT_DEFINED|Unsigned|7|2|1|0|3|3||
|**XBR_EBI_MODE_IS_COMMANDING<sup>4</sup>**|Unsigned|47|1|1|0|0|1||
|**XBR_PRIORITY_IS_COMMANDING<sup>4</sup>**|Unsigned|48|1|1|0|0|1||
|**XBR_CONTROL_MODE_IS_COMMANDING<sup>4</sup>**|Unsigned|49|1|1|0|0|1||
|**NOT_DEFINED_IS_COMMANDING<sup>4</sup>**|Unsigned|50|1|1|0|0|1||

<sup>1</sup> 00=NO_ENDURANCE_BRAKE_INTEGRATION_ALLOWED, 01=ONLY_ENDURANCE_BRAKES_ALLOWED, 10=ENDURANCE_BRAKE_INTEGRATION_ALLOWED, 11=NOT_AVAILABLE

<sup>2</sup> 00=HIGHEST_PRIORITY, 01=HIGH_PRIORITY, 10=MEDIUM_PRIORITY, 11=LOW_PRIORITY

<sup>3</sup> 00=OVERRIDE_DISABLE, 01=ACCELERATION_CONTROL_WITH_ADDITION_MODE, 10=ACCELERATION_CONTROL_WITH_MAXIMUM_MODE, 11-NOT_AVAILABLE

**<sup>4</sup> 0=IS_NOT_COMMANDING, 1=IS_COMMMANDING**

asaxsma, 2019-06-28:
Pull out "not defined".
Remove is commanding.
Add vehicle feedback in aux message.
Make bits part of command message.

asaxsma, 2019-06-28:
Requirement: Command air brakes.
Message Name: BRAKE_DECCEL_CMD
Message ID (Decimal): 312
Message ID (Hexadecimal): 0x138
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLE|Unsigned|0|1|1|0|0|1||
|IGNORE_OVERRIDES|Unsigned|1|1|1|0|0|1||
|CLEAR_OVERRIDE|Unsigned|2|1|1|0|0|1||
|BRAKE_DECCEL_CMD|Unsigned|15|16|0.001|0|0|10|m/s^2|
|XBR_EBI_MODE<sup>1</sup>|Unsigned|25|2|1|0|0|2||
|XBR_PRIORITY<sup>2</sup>|Unsigned|27|2|1|0|0|3||
|XBR_CONTROL_MODE<sup>3</sup>|Unsigned|29|2|1|0|0|2||

<sup>1</sup> 00=NO_ENDURANCE_BRAKE_INTEGRATION_ALLOWED, 01=ONLY_ENDURANCE_BRAKES_ALLOWED, 10=ENDURANCE_BRAKE_INTEGRATION_ALLOWED, 11=NOT_AVAILABLE

<sup>2</sup> 00=HIGHEST_PRIORITY, 01=HIGH_PRIORITY, 10=MEDIUM_PRIORITY, 11=LOW_PRIORITY

<sup>3</sup> 00=OVERRIDE_DISABLE, 01=ACCELERATION_CONTROL_WITH_ADDITION_MODE, 10=ACCELERATION_CONTROL_WITH_MAXIMUM_MODE, 11-NOT_AVAILABLE

Message Name: BRAKE_DECCEL_RPT
Message ID (Decimal): 568
Message ID (Hexadecimal): 0x238
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLED|Unsigned|0|1|1|0|0|1||
|OVERRIDE_ACTIVE|Unsigned|1|1|1|0|0|1||
|COMMAND_OUTPUT_FAULT|Unsigned|2|1|1|0|0|1||
|INPUT_OUTPUT_FAULT|Unsigned|3|1|1|0|0|1||
|OUTPUT_REPORTED_FAULT|Unsigned|4|1|1|0|0|1||
|PACMOD_FAULT|Unsigned|5|1|1|0|0|1||
|VEHICLE_FAULT|Unsigned|6|1|1|0|0|1||
|COMMAND_TIMEOUT|Unsigned|7|1|1|0|0|1||
|MANUAL_INPUT|Unsigned|15|16|0.001|0|0|10|m/s^2|
|COMMANDED_VALUE|Unsigned|31|16|0.001|0|0|10|m/s^2|
|OUTPUT_VALUE|Unsigned|47|16|0.001|0|0|10|m/s^2|

Message Name: BRAKE_DECCEL_AUX_RPT
Message ID (Decimal): 824
Message ID (Hexadecimal): 0x338
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|XBR_EBI_MODE<sup>1</sup>|Unsigned|1|2|1|0|0|2||
|XBR_PRIORITY<sup>2</sup>|Unsigned|3|2|1|0|0|3||
|XBR_CONTROL_MODE<sup>3</sup>|Unsigned|5|2|1|0|0|2||
|XBR_EBI_MODE_IS_VALID|Unsigned|8|2|1|0|0|2||
|XBR_PRIORITY_IS_VALID|Unsigned|9|2|1|0|0|3||
|XBR_CONTROL_MODE_IS_VALID|Unsigned|10|2|1|0|0|2||

<sup>1</sup> 00=NO_ENDURANCE_BRAKE_INTEGRATION_ALLOWED, 01=ONLY_ENDURANCE_BRAKES_ALLOWED, 10=ENDURANCE_BRAKE_INTEGRATION_ALLOWED, 11=NOT_AVAILABLE

<sup>2</sup> 00=HIGHEST_PRIORITY, 01=HIGH_PRIORITY, 10=MEDIUM_PRIORITY, 11=LOW_PRIORITY

<sup>3</sup> 00=OVERRIDE_DISABLE, 01=ACCELERATION_CONTROL_WITH_ADDITION_MODE, 10=ACCELERATION_CONTROL_WITH_MAXIMUM_MODE, 11-NOT_AVAILABLE

asaxsma, 2019-06-28:
Please approve new messages.

JWhitleyAStuff, 2019-06-28:
Approved.

driscoll85, 2019-06-28:
Approved

asaxsma, 2019-07-03:
@mlemm99 do you approve?

mlemm99, 2019-07-03:
Not yet approved. The deccel command is transmitted at 50hz

asaxsma, 2019-07-03:
Leave the report and aux report at 30hz?

JWhitleyAStuff, 2019-07-03:
There is no reason for us to have it received on the bus at 50Hz. The PACMod operates at 30Hz. It can re-send a value to the vendor hardware occasionally.

mlemm99, 2019-07-03:
approved

asaxsma, 2019-07-16:
The following aux report change was requested.  Please approve.

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|XBR_ACTIVE_CONTROL_MODE<sup>1</sup>|Unsigned|3|4|1|0|0|3||
|XBR_SYSTEM_STATE<sup>2</sup>|Unsigned|5|2|1|0|0|2||
|FOUNDATION_BRAKE_USE<sup>3</sup>|Unsigned|7|2|1|0|0|1||
|HILL_HOLDER_MODE<sup>4</sup>|Unsigned|10|3|1|0|0|6||
|XBR_ACTIVE_CONTROL_MODE_IS_VALID|Unsigned|16|1|1|0|0|1||
|XBR_SYSTEM_STATE_IS_VALID|Unsigned|17|1|1|0|0|1||
|FOUNDATION_BRAKE_USE_IS_VALID|Unsigned|18|1|1|0|0|1||
|HILL_HOLDER_MODE_IS_VALID|Unsigned|19|1|1|0|0|1||

<sup>1</sup> 0=NO_BRAKE_DEMAND, 1=DRIVERS_BRAKE_DEMAND, 2=ADDITION_MODE_OF_XBR_ACCELERATION_CONTROL, 3=MAXIMUM_MODE_OF_XBR_ACCELERATION_CONTROL

<sup>2</sup> 0=ANY_EXTERNAL_BRAKE_DEMAND_WILL_BE_ACCEPTED, 2=NO_EXTERNAL_BRAKE_DEMAND_WILL_BE_ACCEPTED

<sup>3</sup> 0=FOUNDATION_BRAKES_NOT_IN_USE, 1=FOUNDATION_BRAKES_IN_USE

<sup>4</sup> 0=INACTIVE, 1=ACTIVE, 2=ACTIVE_BUT_INACTIVE_SOON, 6=ERROR

JWhitleyAStuff, 2019-07-16:
Approved.

mlemm99, 2019-07-16:
approved

driscoll85, 2019-07-16:
approved

closed 2019-08-30 by asaxsma