astuff/pacmod_dbc #45: Fix Shift AUX Report

mlemm99, 2019-07-21:
## Overview
With the shorter DLC, modify the shift aux report to use less bytes as well as follow the flag / valid flag standard

## Current
Message Name: SHIFT_AUX_RPT
Message ID (Decimal): 808
Message ID (Hexadecimal): 0x328
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|BETWEEN_GEARS|Unsigned|0|1|1|0|0|1||
|STAY_IN_NEUTRAL_MODE|Unsigned|1|1|1|0|0|1||
|BRAKE_INTERLOCK_ACTIVE|Unsigned|2|1|1|0|0|1||
|SPEED_INTERLOCK_ACTIVE|Unsigned|3|1|1|0|0|1||
|BETWEEN_GEARS_IS_VALID|Unsigned|8|1|1|0|0|1||
|STAY_IN_NEUTRAL_MODE_IS_VALID|Unsigned|9|1|1|0|0|1||
|BRAKE_INTERLOCK_ACTIVE_IS_VALID|Unsigned|10|1|1|0|0|1||
|SPEED_INTERLOCK_ACTIVE_IS_VALID|Unsigned|11|1|1|0|0|1||
|WRITE_TO_CONFIG|Unsigned|56|1|1|0|0|1||

## New
Message Name: SHIFT_AUX_RPT
Message ID (Decimal): 808
Message ID (Hexadecimal): 0x328
Frame Format: Standard
Data Length Code: 2
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|BETWEEN_GEARS|Unsigned|0|1|1|0|0|1||
|STAY_IN_NEUTRAL_MODE|Unsigned|1|1|1|0|0|1||
|BRAKE_INTERLOCK_ACTIVE|Unsigned|2|1|1|0|0|1||
|SPEED_INTERLOCK_ACTIVE|Unsigned|3|1|1|0|0|1||
|WRITE_TO_CONFIG|Unsigned|4|1|1|0|0|1||
|BETWEEN_GEARS_IS_VALID|Unsigned|8|1|1|0|0|1||
|STAY_IN_NEUTRAL_MODE_IS_VALID|Unsigned|9|1|1|0|0|1||
|BRAKE_INTERLOCK_ACTIVE_IS_VALID|Unsigned|10|1|1|0|0|1||
|SPEED_INTERLOCK_ACTIVE_IS_VALID|Unsigned|11|1|1|0|0|1||
|WRITE_TO_CONFIG_IS_VALID|Unsigned|12|1|1|0|0|1||

JWhitleyAStuff, 2019-07-21:
I approve. However, I think it might be a good idea for us to make a release with the most recent changes before we actually start to implement the shorter DLC thing. That way, we can point to a definitive cut-off point for customers as the last "all-8-DLC" release.

mlemm99, 2019-07-21:
Works for me

driscoll85, 2019-07-21:
Approved

Sent from my Verizon, Samsung Galaxy smartphone

asaxsma, 2019-07-24:
@mlemm99 these fields haven't been deployed to any of the platforms yet?

mlemm99, 2019-07-24:
any platforms currently using bit 56 are still using 56. moving things in line with the flag / flag valid and reducing the bytes has not been deployed yet

asaxsma, 2019-08-19:
## Overview
With the shorter DLC, modify the shift aux report to use less bytes as well as follow the flag / valid flag standard

## Current
Message Name: SHIFT_AUX_RPT
Message ID (Decimal): 808
Message ID (Hexadecimal): 0x328
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|BETWEEN_GEARS|Unsigned|0|1|1|0|0|1||
|STAY_IN_NEUTRAL_MODE|Unsigned|1|1|1|0|0|1||
|BRAKE_INTERLOCK_ACTIVE|Unsigned|2|1|1|0|0|1||
|SPEED_INTERLOCK_ACTIVE|Unsigned|3|1|1|0|0|1||
|BETWEEN_GEARS_IS_VALID|Unsigned|8|1|1|0|0|1||
|STAY_IN_NEUTRAL_MODE_IS_VALID|Unsigned|9|1|1|0|0|1||
|BRAKE_INTERLOCK_ACTIVE_IS_VALID|Unsigned|10|1|1|0|0|1||
|SPEED_INTERLOCK_ACTIVE_IS_VALID|Unsigned|11|1|1|0|0|1||
|WRITE_TO_CONFIG|Unsigned|56|1|1|0|0|1||

## New
Message Name: SHIFT_AUX_RPT
Message ID (Decimal): 808
Message ID (Hexadecimal): 0x328
Frame Format: Standard
Data Length Code: 2
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|BETWEEN_GEARS|Unsigned|0|1|1|0|0|1||
|STAY_IN_NEUTRAL_MODE|Unsigned|1|1|1|0|0|1||
|BRAKE_INTERLOCK_ACTIVE|Unsigned|2|1|1|0|0|1||
|SPEED_INTERLOCK_ACTIVE|Unsigned|3|1|1|0|0|1||
|WRITE_TO_CONFIG|Unsigned|**4**|1|1|0|0|1||
|BETWEEN_GEARS_AVAIL|Unsigned|8|1|1|0|0|1||
|**STAY_IN_NEUTRAL_MODE_AVAIL**|Unsigned|9|1|1|0|0|1||
|**BRAKE_INTERLOCK_ACTIVE_AVAIL**|Unsigned|10|1|1|0|0|1||
|**SPEED_INTERLOCK_ACTIVE_AVAIL**|Unsigned|11|1|1|0|0|1||
|**WRITE_TO_CONFIG_IS_VALID**|**Unsigned**|**12**|**1**|**1**|**0**|**0**|**1**||

-- Andy bolded changed text in the "New" section.

closed 2019-08-30 by asaxsma