astuff/pacmod_dbc #46: Move E-Stop from Component Messages to Miscellaneous Report

asaxsma, 2019-07-29:
Requirement: Move E-Stop from Component Report to Miscellaneous Report for architecture improvement.

## Modify existing message
Message Nam: COMPONENT_RPT_00 through COMPONENT_RPT_03
Message ID (Decimal): 32-35
Message ID (Hexadecimal): 0x20-0x23
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Comment|
|---|---|---|---|---|---|---|---|---|
|COMPONENT_TYPE|Unsigned|3|4|1|0|0|15||
|ACCEL|Unsigned|4|1|1|0|0|1||
|BRAKE|Unsigned|5|1|1|0|0|1||
|CRUISE_CONTROL_BUTTONS|Unsigned|6|1|1|0|0|1||
|DASH_CONTROLS_LEFT|Unsigned|7|1|1|0|0|1||
|DASH_CONTROLS_RIGHT|Unsigned|8|1|1|0|0|1||
|HAZARD_LIGHTS|Unsigned|9|1|1|0|0|1||
|HEADLIGHT|Unsigned|10|1|1|0|0|1||
|HORN|Unsigned|11|1|1|0|0|1||
|MEDIA_CONTROLS|Unsigned|12|1|1|0|0|1||
|PARKING_BRAKE|Unsigned|13|1|1|0|0|1||
|SHIFT|Unsigned|14|1|1|0|0|1||
|SPRAY|Unsigned|15|1|1|0|0|1||
|STEERING|Unsigned|16|1|1|0|0|1||
|TURN|Unsigned|17|1|1|0|0|1||
|WIPER|Unsigned|18|1|1|0|0|1||
|WATCHDOG|Unsigned|19|1|1|0|0|1||
|BRAKE_DECCEL|Unsigned|20|1|1|0|0|1||
|COUNTER|Unsigned|35|4|1|0|0|15||
|COMPLEMENT|Unsigned|39|4|1|0|0|15||
|CONFIG_FAULT|Unsigned|40|1|1|0|0|1||
|CAN_TIMEOUT_FAULT|Unsigned|41|1|1|0|0|1||
|~~ESTOP~~|~~Unsigned~~|~~42~~|~~1~~|~~1~~|~~0~~|~~0~~|~~1~~||
|INTERNAL_SUPPLY_VOLTAGE_FAULT|Unsigned|**42**|1|1|0|0|1||
|~~ESTOP_FAULT~~|~~Unsigned~~|~~44~~|~~1~~|~~1~~|~~0~~|~~0~~|~~1~~||

## Add New Message
Message Nam: E_STOP_RPT
Message ID (Decimal): 1052
Message ID (Hexadecimal): 0x41C
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Comment|
|---|---|---|---|---|---|---|---|---|
|ESTOP<sup>1</sup>|Unsigned|0|1|1|0|0|1||
|ESTOP_FAULT<sup>2</sup>|Unsigned|1|1|1|0|0|1||

<sup>1</sup> 0=NOT_ACTIVE, 1=ACTIVE
<sup>2</sup> 0=NO_FAULT, 1=FAULT

asaxsma, 2019-07-29:
Per our meeting, let's approve this online.   Thanks!

asaxsma, 2019-07-29:
Let's change name from E_STOP_RPT to ESTOP_RPT.

asaxsma, 2019-07-29:
Ope, one more change, let's go with the already approved values for ESTOP of 0 RELEASED 1 PRESSED.

JWhitleyAStuff, 2019-07-29:
I approve.

driscoll85, 2019-07-29:
> This email comes from outside the organization.

Approved

Sent from my Verizon, Samsung Galaxy smartphone

mlemm99, 2019-07-30:
we may want to change the name of the report from estop report to something more generic so we can fill in more bits in the future

JWhitleyAStuff, 2019-07-30:
@mlemm99 - Since we don't have a designated future use, I would say that this is fine for now. We can always rename it (doesn't change the API) and add more data.

closed 2019-08-30 by asaxsma