astuff/pacmod_dbc #50: To Assign a System based Command CAN ID, report CAN ID and aux report CAN ID for a DoorSystem

sanazfth, 2019-08-14:
## To Assign a System based Command CAN Message for a DoorSystem
Requirement:  To Assign a System based Command CAN ID for a DoorSystem
Message Name: DOOR_CMD
Message ID (Decimal): 320
Message ID (Hexadecimal): 0x140
Frame Format: Standard
Data Length Code: 1
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLE|Unsigned|0|1|1|0|0|1||
|IGNORE_OVERRIDE|Unsigned|1|1|1|0|0|1||
|CLEAR_OVERRIDE|Unsigned|2|1|1|0|0|1||
|DOOR_CONTROL_CMD|Unsigned|7|2|1|0|0|2||

1 0=OPEN, 1=NEUTRAL, 2=CLOSE

## To Assign a System based report CAN Message  for a DoorSystem
Requirement:  To Assign a System based Command CAN ID for a DoorSystem
Message Name: DOOR_RPT
Message ID (Decimal): 576
Message ID (Hexadecimal): 0x240
Frame Format: Standard
Data Length Code: 4
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLED|Unsigned|0|1|1|0|0|1||
|OVERRIDE_ACTIVE|Unsigned|1|1|1|0|0|1||
|COMMAND_OUTPUT_FAULT|Unsigned|2|1|1|0|0|1||
|INPUT_OUTPUT_FAULT|Unsigned|3|1|1|0|0|1||
|OUTPUT_REPORTED_FAULT|Unsigned|4|1|1|0|0|1||
|PACMOD_FAULT|Unsigned|5|1|1|0|0|1||
|VEHICLE_FAULT|Unsigned|6|1|1|0|0|1||
|COMMAND_TIMEOUT|Unsigned|7|1|1|0|0|1||
|MANUAL_INPUT|Unsigned|15|8|1|0|0|3||
|COMMANDED_VALUE|Unsigned|23|8|1|0|0|3||
|OUTPUT_VALUE|Unsigned|31|8|1|0|0|3||

## To Assign a System based aux report CAN Message for a DoorSystem
Requirement:  To Assign a System based Command CAN ID for a DoorSystem
Message Name: DOOR_Aux_RPT
Message ID (Decimal): 832
Message ID (Hexadecimal): 0x340
Frame Format: Standard
Data Length Code: 1
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|DRIVER_FRONT_DOOR|Unsigned|0|1|1|0|0|1||
|PASSENGER_FRONT_DOOR|Unsigned|1|1|1|0|0|1||
|DRIVER_REAR_DOOR|Unsigned|2|1|1|0|0|1||
|PASSENGER_REAR_DOOR|Unsigned|3|1|1|0|0|1||
|TRUNK_DOOR|Unsigned|4|1|1|0|0|1||

1 0=CLOSE, 1=OPEN

JWhitleyAStuff, 2019-08-14:
What is this _CMD intended to control? What exactly are we controlling/reporting with a DoorRpt?

sanazfth, 2019-08-14:
> This email comes from outside the organization.

It’s just opening and closing the door!

JWhitleyAStuff, 2019-08-14:
1. There are a LOT of safety implications with this. 
2. We have a current or planned platform where we can control this function? 
3. I can't personally approve of this until we've had some discussions about the above 2.

sanazfth, 2019-08-14:
Well, we are going to have this feature in JPN_TAXI and seems like we use to have a DoorSystem for Audi, which idk much about that. about the safety, you can’t send any command to open or close the door when you are driving, in JPN_taxi and we are just controlling a sliding door.

asaxsma, 2019-08-19:
## To Assign a System based Command CAN Message for a DoorSystem
Requirement:  To Assign a System based Command CAN ID for a DoorSystem
Message Name: REAR_PASS_DOOR_CMD
Message ID (Decimal): 320
Message ID (Hexadecimal): 0x140
Frame Format: Standard
Data Length Code: 2
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLE|Unsigned|0|1|1|0|0|1||
|IGNORE_OVERRIDE|Unsigned|1|1|1|0|0|1||
|CLEAR_OVERRIDE|Unsigned|2|1|1|0|0|1||
|DOOR_CONTROL_CMD|Unsigned|15|8|1|0|0|2||

1 1=OPEN, 0=NEUTRAL, 2=CLOSE

## To Assign a System based report CAN Message  for a DoorSystem
Requirement:  To Assign a System based Command CAN ID for a DoorSystem
Message Name: REAR_PASS_DOOR_RPT
Message ID (Decimal): 576
Message ID (Hexadecimal): 0x240
Frame Format: Standard
Data Length Code: 4
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|ENABLED|Unsigned|0|1|1|0|0|1||
|OVERRIDE_ACTIVE|Unsigned|1|1|1|0|0|1||
|COMMAND_OUTPUT_FAULT|Unsigned|2|1|1|0|0|1||
|INPUT_OUTPUT_FAULT|Unsigned|3|1|1|0|0|1||
|OUTPUT_REPORTED_FAULT|Unsigned|4|1|1|0|0|1||
|PACMOD_FAULT|Unsigned|5|1|1|0|0|1||
|VEHICLE_FAULT|Unsigned|6|1|1|0|0|1||
|COMMAND_TIMEOUT|Unsigned|7|1|1|0|0|1||
|MANUAL_INPUT|Unsigned|15|8|1|0|0|2||
|COMMANDED_VALUE|Unsigned|23|8|1|0|0|2||
|OUTPUT_VALUE|Unsigned|31|8|1|0|0|2||

1 1=OPEN, 0=NEUTRAL, 2=CLOSE

closed 2019-08-30 by asaxsma