astuff/pacmod_dbc #51: Add DoorSystem to componentReport

sanazfth, 2019-08-14:
#Add DoorSystem to componentReport
Requirement: Add DoorSystem to componentReport
Message Name: 
Message ID (Decimal): 32
Message ID (Hexadecimal): 0x20
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|COMPONENT_TYPE|Unsigned|3|4|1|0|0|2||
|ACCEL|Unsigned|4|1|1|0|0|1||
|BRAKE|Unsigned|5|1|1|0|0|1||
|CRUSE|Unsigned|6|1|1|0|0|1||
|DASH_LEFT|Unsigned|7|1|1|0|0|1||
|DASH_RIGHT|Unsigned|8|1|1|0|0|1||
|HAZARD|Unsigned|9|1|1|0|0|1||
|HEADLIGHT|Unsigned|10|1|1|0|0|1||
|HORN|Unsigned|11|1|1|0|0|1||
|MEDIA|Unsigned|12|1|1|0|0|1||
|PARKING_BRAKE|Unsigned|13|1|1|0|0|1||
|SHIFT|Unsigned|14|1|1|0|0|1||
|SPRAY|Unsigned|15|1|1|0|0|1||
|STEER|Unsigned|16|1|1|0|0|1||
|TURN|Unsigned|17|1|1|0|0|1||
|WIPER|Unsigned|18|1|1|0|0|1||
|WATCHDOG|Unsigned|19|1|1|0|0|1||
|DOOR|Unsigned|20|1|1|0|0|1||
|STUFF1|Unsigned|23|3|1|0|0|||
|STUFF2|Unsigned|31|8|1|0|0|||
|COUNTER|Unsigned|35|4|1|0|0|||
|COMPLEMENT|Unsigned|39|4|1|0|0|||
|CONFIG_FAULT|Unsigned|40|1|1|0|0|1||
|CAN_FAULT|Unsigned|41|1|1|0|0|1||
|ESTOP|Unsigned|42|1|1|0|0|1||
|INTERNAL_SUPPLY_VOLTAGE_VOLTAGE_FAULT|Unsigned|43|1|1|0|0|1||
|ESTOP_FAULT|Unsigned|44|1|1|0|0|1||
|STUFF3|Unsigned|47|1|1|0|0|1||

1 0=PACMOD, 1=PACMINI, 2=PACMICRO

asaxsma, 2019-08-19:
#Add DoorSystem to componentReport
Requirement: Add DoorSystem to componentReport
Message Name: 
Message ID (Decimal): 32
Message ID (Hexadecimal): 0x20
Frame Format: Standard
Data Length Code: 8
TX Node: PACMOD
Message Rate: 30 Hz

|Name|Type|Bitpos|Length|Factor|Offset|Min|Max|Unit|
|---|---|---|---|---|---|---|---|---|
|COMPONENT_TYPE|Unsigned|3|4|1|0|0|2||
|ACCEL|Unsigned|4|1|1|0|0|1||
|BRAKE|Unsigned|5|1|1|0|0|1||
|**CRUISE**|Unsigned|6|1|1|0|0|1||
|DASH_LEFT|Unsigned|7|1|1|0|0|1||
|DASH_RIGHT|Unsigned|8|1|1|0|0|1||
|HAZARD|Unsigned|9|1|1|0|0|1||
|HEADLIGHT|Unsigned|10|1|1|0|0|1||
|HORN|Unsigned|11|1|1|0|0|1||
|MEDIA|Unsigned|12|1|1|0|0|1||
|PARKING_BRAKE|Unsigned|13|1|1|0|0|1||
|SHIFT|Unsigned|14|1|1|0|0|1||
|SPRAY|Unsigned|15|1|1|0|0|1||
|STEER|Unsigned|16|1|1|0|0|1||
|TURN|Unsigned|17|1|1|0|0|1||
|WIPER|Unsigned|18|1|1|0|0|1||
|WATCHDOG|Unsigned|19|1|1|0|0|1||
|**REAR_PASS_DOOR**|Unsigned|20|1|1|0|0|1||
|COUNTER|Unsigned|35|4|1|0|0|||
|COMPLEMENT|Unsigned|39|4|1|0|0|||
|CONFIG_FAULT|Unsigned|40|1|1|0|0|1||
|CAN_FAULT|Unsigned|41|1|1|0|0|1||
|INTERNAL_SUPPLY_VOLTAGE_VOLTAGE_FAULT|Unsigned|42|1|1|0|0|1||

1 0=PACMOD, 1=PACMINI, 2=PACMICRO

mlemm99, 2019-08-20:
modified last comment to include the BRAKE_DECCEL system @ bit position 20

closed 2019-08-30 by asaxsma